PX4/flight_review #4: Subsample Google Maps KML track

LorenzMeier, 2016-12-09:
@bkueng I think you need to subsample the KML track to 5 Hz or lower as the current data seems to overwhelm Google. The map is often broken because of it.

bkueng, 2016-12-12:
It's a bug in the bokeh, it does not handle google maps so well. I'll update the library and see if that helps.

LorenzMeier, 2016-12-12:
I'm not sure its bokeh. I've seen this before. I'm pretty confident that you need to sub-sample the data before giving it to Google Maps. Either within bokeh or please use KML and Google maps directly.

This need to work *now* and its not ok to wait for Bokeh to fix it.

*The tool as it stands is useless without the Google maps plots.*

LorenzMeier, 2016-12-12:
Update: I was wrong about the number of points, its a genuine bug in Bokeh requiring a workaround by using plain Google Maps and KML.

LorenzMeier, 2016-12-13:
Opening a new issue for this.

closed 2016-12-13 by LorenzMeier