PX4/flight_review #12: Maps view broken

LorenzMeier, 2016-12-13:
This is from the SITL system, completely unreadable:

[image]

Zooming in and panning even worse:

[image]

LorenzMeier, 2016-12-13:
This log before applying any zooming:
http://logs.px4.io/plot_app?log=e322bc62-56e4-48e9-8dd3-136e2799e789

[image]

bkueng, 2016-12-16:
fixed in #14

closed 2016-12-16 by bkueng