PX4/flight_review #22: Altitude control mode color

dagar, 2017-02-15:
It's very difficult to see the dark gray altitude control section over a satellite map.

[image]

[image]

http://logs.px4.io/plot_app?log=4011b562-64a3-494e-8685-782414499d52

bkueng, 2017-02-22:
Not easy to find another distinct color. How about this:

[image]
[image]

dagar, 2017-02-22:
Anything discernable over a satellite map works for me. Thanks!

bkueng, 2017-02-22:
Done in https://github.com/PX4/flight_review/commit/7d385b03f1e4dc5700f3165b5a5791545acb02cc

closed 2017-02-22 by bkueng