PX4/flight_review #23: Plot vision topics

mhkabir, 2017-02-16:
@bkueng Now that the firmware side is in, we need plotting for vision topics

There should be new plots : 
1. External position, velocity (`vehicle_vision_position` and relevant fields from `vehicle_local_position_groundtruth`)
2. External atttiude, angular velocity (`vehicle_vision_attitude` and relevant fields from `vehicle_attitude_groundtruth`)

It would also be great if you can add ground-truth to the existing local position / velocity plots. I will add plumbing the Firmware side to log data from ATT_POS_MOCAP and in the future from https://github.com/mavlink/mavlink/issues/660 into `vehicle_groundtruth_position

mhkabir, 2017-02-20:
@bkueng @jgoppert @LorenzMeier I tried implementing the vision plotting locally, but there is some sort of an issue.

Here's the diff : 
```
+    # External position
+    data_plot = DataPlot(data, plot_config, 'vehicle_vision_attitude',
+                         y_axis_label='[m]', title='Vision Position',
+                         plot_height='small', changed_params=changed_params)
+    data_plot.add_graph(['x', 'y', 'z'], colors3, ['x', 'y', 'z'])
+    plot_flight_modes_background(data_plot.bokeh_plot, flight_mode_changes)
+
+    if data_plot.finalize() is not None: plots.append(data_plot)
+

```

Here's the console log:
```
kabir@Kabir-MacBookPro:~/PX4/flight_review$ ./serve.py -f ~/.ros/rootfs/fs/microsd/log/2017-02-20/08_18_38.ulg 
Created new window in existing browser session.
GET[log]=/home/kabir/.ros/rootfs/fs/microsd/log/2017-02-20/08_18_38.ulg
<class 'IndexError'> (vehicle_gps_position): list index out of range
<class 'IndexError'> (vehicle_gps_position): list index out of range
<class 'IndexError'> (vehicle_vision_position): list index out of range
<class 'IndexError'> (rc_channels): list index out of range
<class 'IndexError'> (distance_sensor): list index out of range
<class 'IndexError'> (vehicle_gps_position): list index out of range
<class 'IndexError'> (vehicle_gps_position): list index out of range
<class 'IndexError'> (input_rc): list index out of range
<class 'IndexError'> (cpuload): list index out of range
```

Any ideas on these errors? I can open and plot the fields just fine in Flight Plot.

bkueng, 2017-02-22:
You need to add the topic to the list for it to be loaded: https://github.com/PX4/flight_review/blob/master/plot_app/helper.py#L214
Then it should work. The other errors just show you that these topics are not found in the log and thus will not be plotted. It's best effort: it plots what it finds.
The ground truth is already plotted if it's in the log.

Can you provide a log file?

mhkabir, 2017-02-22:
Ah, awesome. Thanks. 
I'll just send in a PR with the stuff implemented, I have it covered now, thanks.

I'll send you a logfile (with only basic vision data from SITL) in a minute.

bkueng, 2017-02-22:
Great, Thanks!

mhkabir, 2017-02-23:
Done!

Log with vision data : http://logs.px4.io/plot_app?log=2af42118-a9e3-42db-826e-8713c6beddc3

closed 2017-02-23 by mhkabir

bkueng, 2017-02-23:
👍 now I expect vision performance to improve significantly :P