PX4/flight_review #34: Upload broken

tops4u, 2017-05-14:
Tried to upload a log today, there is some Problem, as it always returns HTTP 500 on logs.px4.io. 

I don't know who to contact on this issue or if you also operate the px4.io infrastructure.

bkueng, 2017-05-15:
Can you upload the log somewhere & share it? I think your log file is corrupt. You can also try to open it with [pyulog](https://github.com/PX4/pyulog):
```
sudo pip install pyulog
ulog_info <file>.ulg
```

tops4u, 2017-05-15:
Tried the log from here: https://github.com/PX4/Firmware/issues/7237 but it did not work

bkueng, 2017-05-15:
This log file is indeed corrupt: it contains 0 bytes where it should not. Did you download it via QGC?

tops4u, 2017-05-15:
You are right... Must have been a Problem with the QGC Download. Now I retried directly by retreiving it from the SD Card and this is the Result: http://logs.px4.io/plot_app?log=ef90710d-083e-4d5b-8a6e-45bb9f057c15

Just on Improvement Notice: Instead of 500 a Message informing about the Logfile being corrupt would be helpful ;-) 

Thanks for your Help.

closed 2017-05-15 by tops4u

bkueng, 2017-05-15:
> Just on Improvement Notice: Instead of 500 a Message informing about the Logfile being corrupt would be helpful ;-)

That's true. Corrupt log files have been very rare so far and https://github.com/PX4/Firmware/pull/7226 should fix the QGC download issues.