PX4/flight_review #36: Display geotags on map

mhkabir, 2017-05-22:
@bkueng It would be great if we can display the geotag points on the map so that we know where the camera was triggered. They are currently logged in the [`camera_capture` message](https://github.com/PX4/Firmware/blob/master/msg/camera_capture.msg).

[image]

TODO : add link to log with capture packets.

mhkabir, 2017-05-22:
This would also help people using suboptimal camera setups which miss triggers which PX4 did command.

bkueng, 2017-05-22:
Good idea. I'll do that when I get a log

mhkabir, 2017-05-22:
@bkueng Here you go : https://drive.google.com/file/d/0B3WIncZkmRb6Y0piUms0RjkyeTQ/view?usp=sharing

bkueng, 2017-05-22:
Done:
[image]

closed 2017-05-22 by bkueng

mhkabir, 2017-05-22:
Amazing, thanks!