PX4/flight_review #39: PX4 1.6.2 log upload broken

mdailey, 2017-06-13:
PX4 version 1.6.2 adds the feature:

- "New .ulg uLog logging format which directly logs uORB topics and offers more accurate time stamping. Supported in FlightPlot"

I tested 1.6.2 with a VTOL QuadRanger. I loaded 1.6.2 firmware, loaded my parameters from 1.6.1, calibrated the sensors etc. and did a few short flights.

The resulting logs load fine in FlightPlot under Windows (I cannot seem to get FlightPlot working on Ubuntu, but that's a different issue).

But when I try to upload to review.px4.io, I get "HTTP Error 400: Failed to parse the file. It is most likely corrupt."

I attach a sample log: [log_8_2017-6-13-08-42-46.zip](https://github.com/PX4/flight_review/files/1071102/log_8_2017-6-13-08-42-46.zip).

The logs are fine in FlightPlot so should be loadable by Flight Review?

bkueng, 2017-06-13:
This log file is indeed broken. Did you download it via QGC? Which QGC version? Can you copy it directly from the SD card and see if that works?

mdailey, 2017-06-13:
@bkueng thanks for your suggestion. Yes I downloaded that file with QGC 3.1.3 on Linux. Following your advice I tried mounting the SD card under Linux and found that it was corrupt (superblock could not even be read). A chkdsk /f under Windows found many lost clusters.

Sorry for the false alarm regarding 1.6.1 vs 1.6.2 ulogs.

Other observations:
- Strange that PX4 was booting without any SD card warning and letting QGC download logs despite a corrupt SD card!
- It seems that FlightPlot's ulog reader is a little more robust than pyulog.

closed 2017-06-13 by mdailey