PX4/flight_review #41: Flight statistics do not include master

LorenzMeier, 2017-06-16:
The current plot on http://review.px4.io/plot_app?stats=1 does include all releases but not master. Given that a lot of testing is against master it would be great to list it in front of all releases.

bkueng, 2017-06-17:
Yes it does (it says 'including development states'). Otherwise there would be almost no logs at all if I only show the logs for a specific version, which is why I added all. This is how it's accounted: logs are added to the previous release, until there's a new RC/beta (which comes from how the git tagging is done).
Alternative suggestions are welcome. Are you saying you want a 'master' version with all logs that are done after the latest version?

BTW thinking about it now, it also includes PR's - there's currently no way to determine that. We can add the branch name to the log for that.

LorenzMeier, 2017-06-18:
I don't see master in the statistics below - I think we need to somehow be able to show an accumulated flight hour number.

[image]

closed 2017-06-21 by LorenzMeier