PX4/flight_review #48: Enable SSL on webinterface

sanderux, 2017-07-13:
@bkueng This service carries sensitive information. I think it should only be provided over SSL. You can generate a free certificate with letsencrypt. An auto-renewal and easy installer is found here: https://certbot.eff.org/all-instructions/

/cc @LorenzMeier

bkueng, 2017-07-14:
I completely agree and it's something I have had on my list for quite some time. Just a matter of actually doing it

bkueng, 2017-07-20:
It wasn't as simple as advertised, but I got it to work: https://logs.px4.io/. I'll add automatic redirects to https after some testing phase.

closed 2017-07-20 by bkueng

sanderux, 2017-07-20:
What issues did you have? On debian apache it was really as easy as advertised when i tried

sanderux, 2017-07-20:
By the way: perhaps it would be good to redirect http to https by default

bkueng, 2017-07-20:
> What issues did you have? On debian apache it was really as easy as advertised when i tried

Well, first I had to open the port for HTTPS on AWS - that one was obvious. Then I had to figure out if possible and how to make it work for multiple DNS names (logs.px4.io & review.px4.io) with a single certificate. But the most troubles I had with is when I tried the certbot nginx method (I'm using it as a reverse proxy), I just got a timeout without further information as to why. I then tried the manual method which worked, and then then nginx method worked as well, but failed to update the config, so I had to do that myself.

> By the way: perhaps it would be good to redirect http to https by default

That's what I meant with:

> I'll add automatic redirects to https after some testing phase.

sanderux, 2017-07-20:
Ah.. ok, thanks, i'll give everything a test too

sanderux, 2017-07-20:
Works fine as far as i've tested

Hi there!

Your uploaded log file with description 'DQ9-AVY Maiden flight 2' is available under:
https://review.px4.io/plot_app?log=eb9956ee-438a-4e9b-8b83-f0547e40cdda

Use the following link to delete the log:
https://review.px4.io/edit_entry?action=delete&log=eb9956ee-438a-4e9b-8b83-f0547e40cdda&token=0fe470040d1f163691ec752f712f69f5