PX4/flight_review #50: Log upload email

sanderux, 2017-07-20:
A few suggested changes to the email sent by Flight Review:
- Add the title to the subject
- Add the description to the body

This will make searching and archiving easier

bkueng, 2017-07-22:
What do you think about that:

subject:
```
Log File uploaded (DJI Flame Wheel F450, Quadrotor x)
```
body:
```
Your uploaded log file is available under:
http://review.px4.io/plot_app?log=xxx

Description: test description
Feedback: test feedback
Vehicle type: Quadrotor
Airframe: DJI Flame Wheel F450, Quadrotor x
Hardware: PX4FMU_V2
Vehicle UUID: 003C00203036510430353333
Software git Hash: ced700adf2448f3ecef7950b2dd62a81fa42a09e (branch: master)
Upload file name: test_log.ulg
```

sanderux, 2017-07-22:
body looks good. 
subject i'm not sure if that is the title but if it is then good

My preference for subject would be:
FlightReview: [title]

bkueng, 2017-07-24:
To clarify: by title you mean the description or the web site title? My example uses the airframe name.

sanderux, 2017-07-24:
Yes, the description i mean

bkueng, 2017-07-24:
Done: https://github.com/PX4/flight_review/commit/35594965486f4d1dde9ed5950a728973d986e549
If no description is provided, the airframe name is used.

closed 2017-07-24 by bkueng

sanderux, 2017-07-24:
Thanks, this helps companies like us with many of the same airframe