PX4/flight_review #51: peoblem of setup server

feilongfl, 2017-07-26:
i setup a server with ubuntu 16.04 lts,but it can not plot anything.in chrome dev console,it show me websocks not work,and console show me   
>list index out of range  
but i don't know where is wrong.can you give me some advice?thanks!

[image]  
[image]

feilongfl, 2017-07-26:
i run server with the command 
```
./serve.py --port 8080 --allow-websocket-origin 10.1.51.135 --num-procs 0 
```

bkueng, 2017-07-26:
Hi @feilongfl 

You can ignore the `list index out of range` errors, it just means that the log does not contain this data.

Can you try with:
```
./serve.py --port 8080 --allow-websocket-origin '*'
```
or:
```
./serve.py --port 8080 --allow-websocket-origin 10.1.51.135:8080
```

feilongfl, 2017-07-26:
@bkueng thank you for reply. I have find the reason, it is my http proxy's error.

closed 2017-07-26 by feilongfl