PX4/flight_review #54: Clarify dt on sensor sampling

LorenzMeier, 2017-08-02:
[image]

How is the delta time (in blue) calculated here? The reason I ask is that we need to make very clear if the system slips that delta or just the logging process. That makes a huge difference. If it is merely the logging process we might consider moving all these quantities to a new, separate plot and do one for system performance (potentially including control system delay) and one for logging performance.

My primary concern is that people think that the system is unreliable in timing when its actually not.

bkueng, 2017-08-03:
> How is the delta time (in blue) calculated here?

Straight-forward as it's described: the diff of the timestamp between 2 logged samples. In that case the logger drops some samples, which depends on logger priority. The system however did not drop/miss any samples as can be seen from the estimator timeslip. Both curves are just as meaningful, and only together you can infer what's going on. The blue curve is also important for replay, or if the driver sampling runs behind (the timeslip would still stay at 0 then). So to me having them in one place makes very much sense. And you need to know a little bit about the system to draw the right conclusions.

LorenzMeier, 2017-08-03:
Yes, but people generally don't pay attention and the log drops make the estimator time slip curve scale badly on Y. How about we separate them to make this easier to read and less error prone in interpretation?

bkueng, 2017-08-03:
I don't know what you mean by scale badly, it just stays around 0, a clear indicator for a cumulative plot. I don't see how separating helps with that. Another reason why I don't want to add plots is that every added plot increases page load time much more than just adding a graph to an existing plot.

bkueng, 2017-08-07:
I limited the y-axis range and updated the label (https://github.com/PX4/flight_review/commit/d2c57f5f1aa07a5b817df556f1f30cabcdd1b12f). Do you see further improvements?

closed 2024-12-16 by bkueng