PX4/flight_review #57: Feature: Automatic log creation/association to PX4/Firmware issue

TSC21, 2017-08-16:
The idea is that when the user uploads the file, adds the issue description on the `Additional Feedback` field, and then checks an option like `Create Github issue on PX4/Firmware`, popping up a Github login if the user is not logged in yet. I assume that the backbone of the Flight Review would have to include one API like https://github.com/paulirish/git-open or https://github.com/node-gh/gh or similar and the binds between issue creation and the Flight Review capabilities. We could also iterate over this and present options for choosing what plots to show on the Github issue, parameter values, etc.

For example, for `nodegh`, would be something like a script that runs something like:
```bash
gh is --new --title <the_log_description_field> --message <the_additional_feedback_field> --user <the_github_user> --repo <master> --log <the_uploaded_log_file> --plots .... --parameters ...
```

We can add an option for link the flight log with an existent issue also. For example, in `nodegh`, that could be something like:
```bash
gh is <issue_number_or_name> --comment <log_description> --log <the_uploaded_log_file> --others ...
```
In the Flight Review side, we can add an option `Link log to Github isse`, which pops a dropdown list of open issues and/or having the option to the user write the issue number, though I prefer the first option.

Note: to keep track of the suggested in https://github.com/PX4/Firmware/pull/7701.

bkueng, 2017-08-17:
I see your point trying to simplify flight issue reporting, and I support that. But I must say I'm not convinced by this. It's considerable work with little benefit.
It duplicates the github issue functionality (at least the UI part, including the issue template), while having less flexibility (what if the user has several logs showing the same issue?).
Often users don't point to single plots, but just provide a log with a description of the faulty behavior (which is fine). We then need to look at the full log in any case to get a complete picture.

I'll keep it open in case someone else wants to work on it.

TSC21, 2017-08-17:
If the user has several logs pointing to the same issue he can basically point the log for that same issue. Beside, the ability of choosing what plots to show is optional, as the user can basically associate the log to the issue without the plots showing on the issue, just a link to the log.
I understand the commitment would be high. I proposed this as a feature but not requesting. If I have some bandwidth I could try to implement it myself. Just wanted this to be tracked as a poasible feature and in case someone else is interested on it.

closed 2018-02-22 by bkueng