PX4/flight_review #58: Access to the private log DB

mtreml, 2017-08-21:
Hello guys, 
Thank you for this amazing tool! Is it possible to **access/download the whole public flight log database** (instead of writing a scraper that pulls it)? This would make it easier to experiment with all the timeseries available so far.

bkueng, 2017-08-22:
Hi
I was thinking about that already. The logs are stored as ulog files on the server together with an sqllite database. So what I would do is to add a JSON (or similar) interface that lets you retrieve all public logs, which then can be used to download the files. But I don't think I have the time to do that shortly, so I'm happy to accept a PR if you want to do that.

bkueng, 2017-09-26:
This got implemented in https://github.com/PX4/flight_review/pull/61

closed 2017-09-26 by bkueng