PX4/flight_review #59: vehicle mode

dagar, 2017-09-05:
Should flight review use vehicle_status.nav_sate instead of commander_state.main_state for the vehicle mode?

bkueng, 2017-09-05:
I was never sure about the semantic differences. Do you know of any? When I started with Flight Review, I just used what logmuncher used.
Looking at it, most of the system seems to use `vehicle_status.nav_state`, so we probably should as well. Do we even need the `commander_state` topic?

dagar, 2017-09-05:
This has blurred a bit, but main_state was the mode the user requested and nav_state was the actual mode. For example if you were in mission and a failsafe triggered only the nav_state would change and if that condition lifted you'd fall back into nav_state = main_state.

Of course that's not consistently applied anymore. Some failsafes (like low battery) now directly change the main_state. The bottom line is that really only the nav_state matters and commander_state would probably be an easy one to drop the next time we're low on FDs.

bkueng, 2017-09-08:
Done in https://github.com/PX4/flight_review/commit/7a6dd8dd19dfebb58dda2b235e2ea3d5f54996bc. We can now remove `commander_state` from the logged topics.

closed 2017-09-08 by bkueng