PX4/flight_review #60: user input plogging (rc_channels, input_rc, manual_control_setpoint)

dagar, 2017-09-05:
Currently flight review plots rc_channels as "Raw Radio Control Inputs" to show user input. This is the calibrated RC input which is then mapped to manual_control_setpoint x,y,z,r, etc.

[image]

In PX4 RC enters the system through input_rc -> rc_channels -> manual_control_setpoint.
The message manual_control_setpoint is what the controllers actually use and it's more helpful for debugging. Additionally it's the only thing logged when using a gamepad/joystick over mavlink instead of RC.

In summary, I think we should be logging and plotting `manual_control_setpoint` instead of `rc_channels`. I can help make the transition if we're in agreement.

bkueng, 2017-09-05:
It depends a bit what you want to debug, but I tend to agree. And since aux and kill switch and others are contained as well, I think it's ok to move it over.

dagar, 2017-09-05:
Great, I'll get manual_control_setpoint added to logger right now, and temporarily leave the others. Once we have more data we can adjust the flight review side. Thanks

bkueng, 2017-09-08:
Done in https://github.com/PX4/flight_review/commit/24ecc582a8f8dd6be59152ede2f302a1ba0c0128. `manual_control_setpoint` is plotted if it exists, otherwise the old behavior is used.

closed 2017-09-08 by bkueng