PX4/flight_review #68: NAN markers

dagar, 2017-11-03:
These are really helper, but there are a few cases where NAN is used normally. For example an IDLE position setpoint will have a lat/lon set to NAN.

bkueng, 2017-11-03:
Can you provide a log? Is it `vehicle_local_position_setpoint.{x,y,z}`? NaN markers are not enabled automatically for all plots, but are explicitly enabled per-graph.

MaEtUgR, 2017-11-05:
In my opinion NaN setpoints can make perfect sense.
As we are discussing this I use NaN setpoints as interface for FlightTasks that want to skip control of a certain state. e.g. if you want to control only velocity but not position you set velocity setpoints to NaN.

bkueng, 2017-11-06:
> In my opinion NaN setpoints can make perfect sense.

Yep, same for me. But we need a way to make it very clear for which messages and fields it's allowed, and what the meaning of NaN is. The most appropriate place for me would be in the uorb message definition. For example `vehicle_local_position_setpoint` does currently not say anything: https://github.com/PX4/Firmware/blob/master/msg/vehicle_local_position_setpoint.msg

MaEtUgR, 2017-11-06:
@bkueng you mean as comments? or is there any formal way?

bkueng, 2017-11-06:
As comments for now, but a more formal way would be even better. A while ago I was plying with the idea to generate a documentation page for all the topics, similar to the parameters. This would then require a more formal definition for all the comments as well.

dagar, 2017-11-06:
We should formalize unit metadata as well.

bkueng, 2017-11-06:
and then use something like [boost units](http://www.boost.org/doc/libs/1_65_0/doc/html/boost_units.html).

dagar, 2017-11-06:
Yes exactly. I was also looking at https://github.com/nholthaus/units

bkueng, 2018-03-13:
Closing as not enough information is provided / everything should be fine.

closed 2018-03-13 by bkueng