PX4/flight_review #69: Setpoint plots should be bars, not lines

bgat, 2017-11-14:
Since setpoints are values that are effective until the next setpoint is issued, a plot of them should be more like a bar graph than a line. With the current graphs, connecting the setpoint updates with lines makes it visually appear as though the setpoint is changing linearly with time between updates.

This characteristic is more readily apparent with flights and/or vehicles that have slow setpoint updates. For example, my vehicle's yaw rate setpoint changes only once a minute or so, and only by large values. The graph plots end look like a sawtooth, rather than a square wave. See the following plot for an example.

[image]

bkueng, 2017-11-14:
It's rather unusual to have such low setpoint updates. But I agree a step graph would be better for that. Can you provide that log, so that I can test what works best?
Alternatively I'll gladly accept a PR.

bgat, 2017-12-04:
How about this log: https://logs.px4.io/plot_app?log=6558436e-1d76-4a6d-8e84-51dc18cdf9af

bkueng, 2017-12-05:
Thanks, that works. I'll take a look at it.

bkueng, 2017-12-06:
After upgrading to the latest bokeh version (#73), this has become really easy: https://github.com/PX4/flight_review/pull/74

closed 2017-12-06 by bkueng