PX4/flight_review #75: 'Acceleration Power Spectral Density' doesn't work on high data rate log

BazookaJoe1900, 2017-12-09:
In case of High rate log the Spectral Density fails and this is how the plot looks like:
[image]

bkueng, 2017-12-11:
What do you mean by high rate log? Can you provide a log?

FYI @joaeba

joaeba, 2017-12-13:
It looks like there could be an issue, but it's hard to debug without a test log. I'll have a look at it if you provide one.

BazookaJoe1900, 2017-12-22:
Sorry for the late replay, here is log produced from the SITL where you can see the issue.
It happens when the SDLG_PROFILE is set to 17 (from the default of 3)
https://logs.px4.io/plot_app?log=357dc188-f02c-4559-8ea3-cd8ebbb600e3
[image]

joaeba, 2017-12-22:
@BazookaJoe1900 ok, thank you. I'll have a look at it.

joaeba, 2017-12-23:
when I open your log file with pyulog, I find your raw sensor data to be sampled only at about 10 Hz (also noticeable when you zoom into the gyro and acceleration plots), while your attitude data is sampled at about 400 Hz. I assume it's a logging problem or could it also be related to pyulog @bkueng ?

BazookaJoe1900, 2017-12-26:
@joaeba You right, the default of 3 on SDLOG_PROFILE sets bits 'DEFAULT' and 'ESTIMATOR_REPLAY'
But setting 17 sets 'DEFAULT' and 'HIGH_RATE' (not 'ESTIMATOR_REPLAY').
The Raw data is on 'sensor_combined' message. that is set to Max rate on 'ESTIMATOR_REPLAY' but to 10Hz at 'DEFAULT'. that why the rate of the Raw data rate is low.
I think you can either close the issue or maybe make the spectral look a bit better on that lower rate of Raw data

bkueng, 2018-01-09:
PR got merged. In addition, I added `sensor_combined` to the high rate profile in https://github.com/PX4/Firmware/pull/8641/commits/0be70896ae7743689fbdaf78bbf77a7ab7f752d1

closed 2018-01-09 by bkueng