PX4/flight_review #77: error: (vehicle_local_position_groundtruth): list index out of range

whyscience, 2017-12-15:
Run locally, got this error.
This is the ulog:
https://review.px4.io/plot_app?log=2f9610e9-87fa-409a-bb8f-0fab6c61971a

```
% ./serve.py -f log_206_2017-12-14-16-22-02.ulg                                                                                                                                                                                 eric@eric-Lenovo
[8157:8196:1215/152545.162167:ERROR:browser_gpu_channel_host_factory.cc(107)] Failed to launch GPU process.
Created new window in existing browser session.
GET[log]=/home/eric/all_code/flight_review/log_206_2017-12-14-16-22-02.ulg
Downloading http://px4-travis.s3.amazonaws.com/Firmware/master/airframes.xml
<class 'IndexError'> (vehicle_local_position_groundtruth): list index out of range
Traceback (most recent call last):
  File "/home/eric/all_code/flight_review/plot_app/main.py", line 185, in <module>
    plots = generate_plots(ulog, px4_ulog, db_data, vehicle_data)
  File "/home/eric/all_code/flight_review/plot_app/configured_plots.py", line 187, in generate_plots
    mark_nan=True, use_step_lines=True)
  File "/home/eric/all_code/flight_review/plot_app/plotting.py", line 467, in add_graph
    p.step(x='timestamp', y=field_name, source=data_source,
AttributeError: 'Figure' object has no attribute 'step'

```

whyscience, 2017-12-15:
also this:
### Internal Server Error

Please open an issue on https://github.com/PX4/flight_review/issues with a link to this log.

joaeba, 2017-12-15:
@whyscience you have to update your bokeh version.

bkueng, 2017-12-15:
> error: (vehicle_local_position_groundtruth): list index out of range

You can ignore this. It just means the log does not contain this topic.

whyscience, 2017-12-15:
@joaeba after I upgrade bokeh, the issue is fixed.

closed 2017-12-15 by whyscience