PX4/flight_review #78: Feature request - Add option to sync all plots time

BazookaJoe1900, 2017-12-26:
Many times you want to look at same time frame at all plots.
It will be nice to have option to do so.

dagar, 2017-12-26:
Interesting idea, although I don't know if it's feasible. Can you expand on how you'd see this working?

Would there be an overall control that applies to all plots? Or perhaps all plots could be updated to the same time scale when one is changed.

joaeba, 2017-12-27:
I assume you talk about zooming just the time axes on the other plots which would generally be feasible. However, I imagine this could lead to a performance problem since at certain zoom levels a lot of data would need to be resampled: currently the data in a plot is dynamically resampled while the user zooms in/out.

BazookaJoe1900, 2017-12-27:
Maybe adding checkbox as "sync time" so only selected plots?
Or, button that if you want to sync to the "time frame". the "time frame" is the time start and end zoom of the last plot that has been changed (it can be written some where on the GUI to be sure)

bkueng, 2018-01-08:
I already thought about this, but have not found a good & simple user experience for that. While fixed linked axis are [relatively simple](https://bokeh.pydata.org/en/latest/docs/user_guide/interaction/linking.html), it's more effort to change them dynamically. The performance impact mentioned by @joaeba is also a concern.