PX4/flight_review #81: Feature request: Show the correct time stamp derived from LOG.

tubeme, 2018-01-11:
The flight review now as Logging Start time show incorrect time of the flight.

It shows the time of the flight + the local time difference (ie. the previewer time difference from the gmt)

It should show the time of the flight as it was in its time zone during the flight.

For example.

I am in UTC+2 zone. I'm the reviewer.

The flight is done in UTC-8 zone.

So when I upload the file to review as Logging Start time it should show UTC-8 and now it shows UTC+2.

It is very hard to work in different time zones this way also trying to sync to a Video of the flight.

bkueng, 2018-01-11:
Neither of them is wrong, so I don't see anything to fix. We can however discuss which one should be used, and I see your point on using log's time. When I implemented it, I followed this reasoning: web services typically show times in your timezone (take calendars for example), so as a user I would expect to see times that are easy to interpret for me (match my own timezone).
I don't feel strongly which one to use (using the browser's timezone is actually more difficult to implement). Maybe we can show both (via tooltip or something)?

tubeme, 2018-01-12:
May be both is good idea... We really have this issue now, flying it two time zones...

closed 2018-01-16 by bkueng