PX4/flight_review #95: Browse http://localhost:5006/browse problems, I have added the data, this is the relationship of windows

Zhenbin24, 2018-03-27:
ERROR:C:\Users\zhen24\AppData\Local\Programs\Python\Python36\lib\site-packages\bokeh\document\document.py:Module <module 'bk_script_7f691323fa25400da
fd8b234f5ab01c9' from 'C:\\Users\\zhen24\\Desktop\\flight_review-master\\flight_review-master\\plot_app\\main.py'> has extra unexpected referrers! Th
is could indicate a serious memory leak. Extra referrers: [(<bokeh.application.handlers.code_runner.CodeRunner object at 0x0000021F108C8A58>, <module
 'bk_script_7f691323fa25400dafd8b234f5ab01c9' from 'C:\\Users\\zhen24\\Desktop\\flight_review-master\\flight_review-master\\plot_app\\main.py'>, <fun
ction CodeHandler.modify_document.<locals>.post_check at 0x0000021F108D5D90>)]

Sincerely ask you ！

MaEtUgR, 2018-03-27:
@Zhenbin24 I think a description of what you're trying to do and what you actually did until now is necessary. Generally flight_review is designed to run on a webserver and together with a database. Did you follow any documenation?

Zhenbin24, 2018-03-27:
@MaEtUgR I haven't made any changes at the moment. I just want to run the program locally. The log of the test is downloaded on your web, Still do not quite understand the problem？

bkueng, 2018-03-27:
@Zhenbin24 Please post your commands that you executed (including installation of packages).

Zhenbin24, 2018-03-28:
@bkueng The above error suddenly disappeared, but there is another problem:
Operation flow: 1. Download the source code 2. Initialize the database 3. Start the program "serve.py -s"
4. Download the blog data from https://review.px4.io/browse and upload it to the local program
5, click browse, no data in the table, so boldly change the public field in the data in the database to 1, or no data and prompt: WARNING: bokeh.server.views.ws: Failed sending message as connection was Closed
6, click Statistics, reported an error IndexError: list index out of range
The detailed errors are as follows:
Error: no generated data
ERROR:bokeh.application.application:Error running application handler <bokeh.application.handlers.director
y.DirectoryHandler object at 0x00000246219D79B0>: list index out of range
File "statistics_plots.py", line 508, in _plot_public_data_statistics:
Prev_version = versions[0] Traceback (most recent call last):
  File "C:\Users\zhen24\AppData\Local\Programs\Python\Python36\lib\site-packages\bokeh\application\handlers\code_runner.py", line 173, in run
    Exec(self._code, module.__dict__)
  File "C:\Users\zhen24\Downloads\flight_review-master\flight_review-master\plot_app\main.py", line 63, in <module>
    p = statistics.plot_public_airframe_statistics()
  File "C:\Users\zhen24\Downloads\flight_review-master\flight_review-master\plot_app\statistics_plots.py", line 414, in plot_public_airframe_statisti
Cs
    Return self._plot_public_data_statistics(self._all_airframes, 'airframes', 'Airframe', label_callback)
  File "C:\Users\zhen24\Downloads\flight_review-master\flight_review-master\plot_app\statistics_plots.py", line 508, in _plot_public_data_statistics
    Prev_version = versions[0]
IndexError: list index out of range

Zhenbin24, 2018-03-29:
@bkueng I modified the upload form and uploaded the public logs. Statistics  Can also display all the log information, including the total number of uploads, etc. Why is the form information for browse still blank? How can we call to display the uploaded record?

closed 2018-04-08 by Zhenbin24