PX4/flight_review #100: Request: Common time marker line among graphs

bigbellmercy, 2018-03-30:
Wow, this flight review software is powerful and professional!
Log analysis efficiency is 200 % increased by it.

If below function is made, it will be the best tool ever.

Function:
1. The Common Time Marker Line displayed as a vertical line on the graph.
2. And side of the line, the time of the line is displayed.
3. If we slide it horizontally, it moves and,
4. All the other lines of other graphs are moved simultaneously to the same time. (fantastic)

Reason:
The root (primary) cause of a problem of flight can be judged when a very small time differences of events among graphs can be discernible.

bkueng, 2018-04-03:
Thanks!

There was already an attempt to implement axis synchronization (https://github.com/PX4/flight_review/pull/89) but had various issues that prevented it from being usable. If you come up with a good way to implement it, please go ahead and do so!
As this is mostly a duplicate of https://github.com/PX4/flight_review/issues/78, I will close this.

closed 2018-04-03 by bkueng

bigbellmercy, 2018-04-03:
Good. I don’t have such skill to implement this function. It is very hard issue to imlement it according to those #78 and #89.