PX4/flight_review #101: Problem: Graph functions does not work after some times pass

bigbellmercy, 2018-03-30:
Problem:
Since all functions of graphs (zoom, pan and etc.) works well in the beginning, after some time passes, those functions does not work for all the graphs. In this status, the buttons are pushed, flight modes are displayed when hovered on graphs and rubber band appears but they does not work.

Workaround:
If the web page reloaded by web browser refresh function, all works well again.

The issue #99 seems to be related this issue.

bkueng, 2018-04-03:
This is either a problem with the internet connection (maybe on your side?), or a problem in the [bokeh library](https://github.com/bokeh/bokeh) that flight review uses. In either case there is not much I can do in flight review.

It could also be that the browser closes the connection due to inactivity, but bokeh has a [keep-alive](https://bokeh.pydata.org/en/latest/docs/reference/command/subcommands/serve.html) option, set to 37 seconds by default. What I can try is reducing this. Does it happen after more than 37 seconds?

bigbellmercy, 2018-04-04:
Yes, it happens more than 37 seconds.
The time seems to be several minutes or a few tens of minutes.
Currently refreshing workaround is likely the solution.

bkueng, 2018-04-04:
Ok then changing the keep-alive option will not help.
Yes, refreshing is the best work-around. Sorry, I don't have a better solution.
Closing this, but should it be a persistent problem to other users as well, we need to revisit.

closed 2018-04-04 by bkueng

bigbellmercy, 2018-04-04:
Good!