PX4/flight_review #104: Altitude setpoint plot is wrong in OFFBOARD mode

anassinator, 2018-04-11:
To preface, I'm not sure if this should be a `Firmware` or `flight_review` issue or both.

# Problem
As it stands, in OFFBOARD mode, sending a [SET_POSITION_TARGET_LOCAL_NED](https://mavlink.io/en/messages/common.html#SET_POSITION_TARGET_LOCAL_NED) with a valid altitude (i.e. `z` coordinate), only the `setpoint_triplet.current.z` is set, and `setpoint_triplet.current.alt` remains 0 ([relevant lines](https://github.com/PX4/Firmware/blob/1b1617b5a0c760119b1cff28db5eceb39626167d/src/modules/mavlink/mavlink_receiver.cpp#L957-L963)). This means that the altitude plot in OFFBOARD mode shows all setpoints to be at 0 altitude despite in reality possibly being something else.

## Example Log
[Log](https://logs.px4.io/plot_app?log=fb6ee64c-32b4-470b-9b5e-3911aaeb13cf)

### Problematic plot
[image]

### Expected plot
[image]

# Possible fixes
I see 2 potential fixes, but not sure either are good.
1. Change the `Firmware` to set `setpoint_triplet.current.alt` when setting `setpoint_triplet.current.z` in OFFBOARD mode. This means logs taken with older firmware will still show the wrong plot though.
2. Fallback to `setpoint_triplet.current.z` when `setpoint_triplet.current.alt = 0` (and maybe only while in OFFBOARD mode) in `flight_review`.

MaEtUgR, 2018-04-12:
@anassinator I think offboard should produce log data compatible to other modes from the Firmware side. Offboard mode is currently under improvement. For better modularization of the different setpoint sources there's now a flight task architecture: https://github.com/PX4/Firmware/pull/8933 and @Stifael is currently working on the task for offboard mode. Chances are he's already aware of that inconsistency.

anassinator, 2018-04-12:
Perfect thanks!

bkueng, 2024-12-16:
Closing, let me know if this is still an issue

closed 2024-12-16 by bkueng