PX4/flight_review #113: UAVCAN ESC Data Plots

keenanjohnson, 2018-06-11:
It would be nice if this review tool showed the data coming back from UAVCAN USCs.

At minimum it should be:

- Voltage
- Current
- Computed Power

MaEtUgR, 2018-06-19:
Could these values not get fed back to the general vehicle power which then is supported by flight review?
Currently the general message does not support distinguishing motors but it might be worth to look into having a general message to capture this information and not specific to UAVCAN because other manufacturers might have similar requirements but different ESCs.

MaEtUgR, 2018-06-27:
That's true, summing up kills too much information. But a general ESC report topic for any ESCs that support feedback instead of one specialized to UAVCAN would still make sense.

You thought about https://github.com/PX4/Firmware/blob/master/msg/esc_status.msg right?

keenanjohnson, 2018-06-27:
No I was referring to the actual ESC Current Draw data which is only reported by CAN bus esc.

For example, something like the snippet below.

```
  # ESC Currents
  data_plot = DataPlot(data, plot_config, 'esc_status',
                       y_axis_label='Current (A)', title='Motor Currents',
                       plot_height='small', changed_params=changed_params,
                       x_range=x_range)
  data_plot.add_graph(['esc[0].esc_current'], colors3, ['mtr1'])
  data_plot.add_graph(['esc[1].esc_current'], colors3, ['mtr2'])
  data_plot.add_graph(['esc[2].esc_current'], colors3, ['mtr3'])
  data_plot.add_graph(['esc[3].esc_current'], colors3, ['mtr4'])
  plot_flight_modes_background(data_plot, flight_mode_changes, vtol_states)

  if data_plot.finalize() is not None: plots.append(data_plot)
```

keenanjohnson, 2018-06-27:
Perhaps we could only display that plot if the ESC type is CAN Bus?

bkueng, 2018-06-28:
My main concern was adding another plot (it increases page load time). But since this topic only applies to tap_esc and uavcan drivers, I'm ok with it.
Can you create a PR and change it so that it checks the esc count `esc_count` and only draws as many lines as there are ESCs?

keenanjohnson, 2018-07-02:
Yeah I'll work on that and have a PR in a few days.

MaEtUgR, 2018-07-10:
Sounds very good, the only point I wanted to make is that it's not only CAN ESCs but also tap_esc and maybe others we don't know yet in the future. Thanks @keenanjohnson .

dagar, 2018-07-10:
> My main concern was adding another plot 

Do you see any relatively easy options for scaling? For example something like sections that only load if you expand them or modes? There are lots of other things I'd like to be able to plot, but not if it burdons existing usage.

bkueng, 2018-07-10:
I tried different things like expanding or click-to-load, but it did not work reliably. Either it was too much javascript for the browser or too much for bokeh.
We can add different pages for specific use-cases. Or try something with iframes then we can expand within the same page.

RicardoM17, 2021-01-28:
@bkueng Were I to revive this, is the situation changed? Do you still share the concern about adding more plots?

I feel like RPM, Current would be the main ones, perhaps also online status but that one will be continuous line in 99.9% of the flights

bkueng, 2021-01-29:
> @bkueng Were I to revive this, is the situation changed? Do you still share the concern about adding more plots?

Yes, there are several options, including a separate page, click-to-load plots or using a different framework.
Adding an RPM plot is fine though.

RicardoM17, 2021-01-29:
> > @bkueng Were I to revive this, is the situation changed? Do you still share the concern about adding more plots?
> 
> Yes, there are several options, including a separate page, click-to-load plots or using a different framework.
> Adding an RPM plot is fine though.

Sounds good. I'll try to find the time to add an RPM plot then.