PX4/flight_review #115: Local Instance of Flight Review Doesn't Accept Uploaded Logs

keenanjohnson, 2018-06-24:
I'm attempting to run the flight review program locally and I can't quite get it to accept any logs.

I'm running the "serve.py" script.

It serves the splash screen fine.

However, when I attempt to upload a file, I get the following:

[image]

keenanjohnson, 2018-06-25:
Actually I was able to resolve this by installing ATLAS

`sudo apt-get install libatlas-base-dev`

Perhaps the readme can be documented to include this dependency?

bkueng, 2018-06-25:
Thanks I'll add it. Are you on ubuntu 18.04?
According to [this](https://packages.ubuntu.com/bionic/amd64/libatlas3-base/filelist), `libatlas3-base` is the right package to install.

keenanjohnson, 2018-06-25:
I was attempting to run this on a raspbian image actually. Perhaps everyone else is running this in ubuntu, so it's not an issue?

bkueng, 2018-06-25:
No, it should work on all (major) distros, that's why I was asking (I'm using it on Fedora for instance).
I updated the instructions in https://github.com/PX4/flight_review/commit/a45774b7a42233e21a6b8ac3327c144e6dc6385a, thanks

closed 2018-06-25 by bkueng