PX4/flight_review #116: Attempting to add custom plot to local instance

keenanjohnson, 2018-06-26:
Hello there.

I'm attempting to add a custom plot to a local instance of this application.

I simply added this section of code to configured_plots.py file, but it doesn't seem to have added the plot.

Does anyone know what I am missing?

data_plot = DataPlot(data, plot_config, 'esc_status_0',
                     y_axis_label='Current (A)', title='Motor Currents',
                     plot_height='small', changed_params=changed_params,
                     x_range=x_range)
data_plot.add_graph(['esc[0].esc_current'], colors3, ['mtr1'])
plot_flight_modes_background(data_plot, flight_mode_changes, vtol_states)

if data_plot.finalize() is not None: plots.append(data_plot)

keenanjohnson, 2018-06-26:
The "data" object doesn't appear to have an object for the esc_status_0 topic, even though I can see it in the flightplot desktop app.

Do I need to add that topic to another subscription list somewhere?

bkueng, 2018-06-26:
Yes you need to add the topic here: https://github.com/PX4/flight_review/blob/master/plot_app/helper.py#L296

keenanjohnson, 2018-06-27:
Got it. Thank you very much!

closed 2018-06-28 by bkueng