PX4/flight_review #117: Add velocity setpoint in web plot of velocity state?

MaEtUgR, 2018-06-27:
Dear @bkueng could we please also get the velocity setpoint ploted in the velocity graph of flight review?
https://review.px4.io/plot_app?log=bcbf6e41-dc35-4291-8711-39e994561a30#Nav-Velocity

Because it's hard to interpret the graph without setpoint and as a result @Stifael and me usually default to downloaing the log and using our profile in flightplot just to analyze velocity.

bkueng, 2018-06-27:
How about this:
[image]
Does it work in the same plot? I'm trying to avoid adding more plots.

MaEtUgR, 2018-06-27:
Looks perfect, the user can always disable lines if he doesn't see the forest anymore or do you mean because of performance? For performance and overwiev the plots could be selective and only the most important ones loaded automatically but that would certainly require bigger changes...

Stifael, 2018-06-27:
another alternative would be to have 3 plots for velocities, but only 1 plot for local position. To me velocity information are more valuable than position. 
But lets just stick with @bkueng suggestion and then go from there if still required

bkueng, 2018-06-27:
Updated in https://github.com/PX4/flight_review/commit/fe116acc1caf336bc9c419be262a149598c66221.

> the user can always disable lines if he doesn't see the forest anymore or do you mean because of performance

Yes, exactly. Adding more plots would increase page load time.

> For performance and overwiev the plots could be selective and only the most important ones loaded automatically but that would certainly require bigger changes

Indeed. I already tried a couple of things in the past, but nothing worked reliably.

> another alternative would be to have 3 plots for velocities, but only 1 plot for local position. To me velocity information are more valuable than position.

Ok, thanks for the feedback, I'll keep that in mind.

closed 2018-06-27 by bkueng

MaEtUgR, 2018-06-27:
❤️

JohannesBrand, 2018-06-27:
@bkueng nitpick: how about changing the Y setpoint color to light green such that it's more or less consistent (red -> orange, dark blue -> light blue, dark green -> then light green). might be that light green would be too salient, though?

bkueng, 2018-06-27:
> @bkueng nitpick: how about changing the Y setpoint color to light green such that it's more or less consistent (red -> orange, dark blue -> light blue, dark green -> then light green). might be that light green would be too salient, though?

Yeah I thought about it but light colors tend to interfere with the flight modes background, so I wanted to use one of the existing 8 colors that I know work. Feel free to play around with it though.