PX4/flight_review #121: Request: search + filters for public logs

dagar, 2018-07-14:
I don't know how viable this is, but I might as well ask.

From public flight reports only I'd like to be able to find all the logs from particular vehicles, airframes, flight controllers, versions of PX4 (specific commits and tags).

 - Vehicle UUID
 - Airframe
 - Hardware
 - Software version

 - basic filters to eliminate irrelevant logs (no takeoff or very short)

bkueng, 2018-07-16:
Except for the last one this is already possible. Check out the search field.

dagar, 2018-07-16:
Yes I tried that first, but assumed it only searched entries in the table. I wasn't able to get any results searching for vehicle UUID or git sha.

bkueng, 2018-07-17:
You're right, looks like we did not have everything upstream yet.

closed 2018-07-28 by bkueng