PX4/flight_review #130: Add TECS plots

RomanBapst, 2018-09-17:
Would be useful to have a few plots for TECS

moreba1, 2019-08-23:
5 days ago 
Hi

it is useful to see airspeed setpoint (airspeed_sp) and filtered airspeed (airspeed_filtered) in airspeed plot .

and there is need to see height_rate and height_rate_setpoint for fixedwings and vtol drones

bkueng, 2019-08-28:
Can you do it and create a pull request? Alternatively send me a log?

moreba1, 2019-08-28:
@bkueng 
could you add this plots?

moreba1, 2019-08-28:
no , i can not do it , sample log :
https://review.px4.io/plot_app?log=b6146ff1-7a47-4259-a6ed-be9f05e1d172

bkueng, 2019-08-29:
I added the Height Rate in https://github.com/PX4/flight_review/commit/6ebb1a59762ee0d30f596d2317f7140730179bcc:

[image]

moreba1, 2019-08-29:
Thank you . if it is possible add airspeed setpoint and filtered airspeed an airspeed tab .

bkueng, 2024-12-16:
We have that now, let me know if anything is missing

closed 2024-12-16 by bkueng