PX4/flight_review #131: Debug of the parsing process? My log ends abruptly

Mtbarros, 2018-09-20:
Hi all,

Sorry if this is the wrong place to discuss this, but I just uploaded a log from a flight from today which ends quite abruptly in the flight review. The log is a bit long because I was doing some testing on the IC engine before flight, and never disarmed and rearmed the plane.

https://review.px4.io/plot_app?log=6cf4fa02-2b5b-4bce-9789-552fa27e550e

It seems suspicious that the long ends right when entering mission mode. There was no impact in flying whatsoever, so I don't think it was a power issue. Also, the additional data is only present for pre-flight, while the post-flight can be seen at the end of the .ulog file.

Can the log be corrupted in the middle and that's why it ends like this? Also, is there any way to debug the parsing process to check for this?

Even if I lost some middle part of the flight it would be great if I could recover the rest of it.

Thank you for your time

bkueng, 2018-09-21:
Hi
the log file is corrupt in the middle - how did you download it? I'd try redownloading directly from the SD card.
I made the parser more robust so that it is able to recover the rest of the file: https://github.com/PX4/pyulog/pull/31.

Mtbarros, 2018-09-21:
Hi, thanks for the response. I downloaded it with USB, and sadly deleted it from the pixhawk before trying to upload. Thank you for the quick help in solving it though. The data is plenty good as it is.

closed 2018-09-21 by Mtbarros