PX4/flight_review #134: 3D representation not possible

hamigolbabaei, 2018-10-09:
you get all the Time this message

"An error occurred while rendering. Rendering has stopped.
RangeError: Invalid array length
RangeError: Invalid array length
    at Ie (https://logs.px4.io/plot_app/static/cesium/Build/Cesium/Cesium.js:544:28871)
    at Re (https://logs.px4.io/plot_app/static/cesium/Build/Cesium/Cesium.js:544:31592)
    at Qe (https://logs.px4.io/plot_app/static/cesium/Build/Cesium/Cesium.js:545:11066)
    at Ye (https://logs.px4.io/plot_app/static/cesium/Build/Cesium/Cesium.js:545:9066)
    at it (https://logs.px4.io/plot_app/static/cesium/Build/Cesium/Cesium.js:545:16460)
    at we.render (https://logs.px4.io/plot_app/static/cesium/Build/Cesium/Cesium.js:545:24412)
    at A.render (https://logs.px4.io/plot_app/static/cesium/Build/Cesium/Cesium.js:557:17428)
    at t (https://logs.px4.io/plot_app/static/cesium/Build/Cesium/Cesium.js:557:5269)“

zanemountcastle, 2018-10-09:
I'm having the same issue, even on 3D visualizations that used to work. In other words, it doesn't seem connected to the individual logs or the particular upload. Instead, the problem seems to be with Cesium.

bkueng, 2018-10-10:
The reason is that the terrain data got deactivated (https://cesiumjs.org/data-and-assets/terrain/stk-world-terrain/) and I'll have to update to the new dataset.

hamigolbabaei, 2018-10-10:
@bkueng Thanks for your feedback, what do you think when I can use the 3D rendering again?

bkueng, 2018-10-10:
Fix is up: https://github.com/PX4/flight_review/pull/135

closed 2018-10-10 by bkueng