PX4/flight_review #136: Feature Request - Add ability to edit flight name and description

deksprime, 2018-10-29:
Changing the flight name and description might be a very useful addition to the flight review functionality. After the log has been uploaded the logs are visually inspected by the uploader and the resulting insights are currently not being annotated within the log page. I think this might be useful for both the uploader and the general database of the flight review.

bkueng, 2018-11-05:
I agree this would be very useful. However the best way to do that and to avoid arbitrary edits is to have user logins, which is a bigger topic.

deksprime, 2018-11-05:
Oh... That's correct, anyone shouldn't be able to edit so I guess this feature is waiting on the user login function implementation.