PX4/flight_review #138: bug: map projections with nan values

mrpollo, 2018-11-01:
In [plot_app/helper.py#227-229](https://github.com/PX4/flight_review/blob/master/plot_app/helper.py#L227-L229)

Consider the following conditionals

```python
    arg = sin_anchor_lat * sin_lat + cos_anchor_lat * cos_lat * cos_d_lon
    arg[arg > 1] = 1
    arg[arg < -1] = -1
```

for the following example value of `arg` of type [`Numpy.Array`](https://docs.scipy.org/doc/numpy-1.15.1/reference/generated/numpy.array.html)

```
array([-0.0488, -0.042 ,  1.    ,  1.    ,  1.    ,     nan,  1.    ,
        1.    ])
```

Looks like we either filter `nan` values or learn how to deal with them

bkueng, 2018-11-05:
Probably filtering them out is best.

mrpollo, 2018-11-05:
I wasn't sure if that was feasible, I don't think I grasp 100% of the math here, and I wasn't sure if filtering out `NaN` would even be a problem.

I'll submit a PR shortly.

bkueng, 2018-11-06:
Can you provide the log(s)? It should not contain NaNs.

mrpollo, 2018-11-06:
I randomly picked a log out of [logs.px4.io/browse](https://logs.px4.io/browse), I lost track of which one it was and since I'm not the original author I won't be re-uploading it to the site, instead here's a [link to download](https://www.dropbox.com/s/e6fumwkcoc6s9n7/ba165f7d-113d-4c5d-b479-1fcf21e99175.ulg?dl=0) that will probably remain available for a few weeks (sorry if you are reading this and the file is no longer available)

bkueng, 2018-11-07:
Thanks, I'll have a look. Log is here btw (the ID was still in the filename): https://logs.px4.io/plot_app?log=ba165f7d-113d-4c5d-b479-1fcf21e99175

bkueng, 2018-11-08:
For the `lat`, `lon` issue, you need to filter earlier, like this:
```diff
diff --git a/plot_app/helper.py b/plot_app/helper.py
index c2018eb..a26184c 100644
--- a/plot_app/helper.py
+++ b/plot_app/helper.py
@@ -218,6 +218,11 @@ def WGS84_to_mercator(lon, lat):

 def map_projection(lat, lon, anchor_lat, anchor_lon):
     """ convert lat, lon in [rad] to x, y in [m] with an anchor position """
+
+    # filter NaN values
+    lat = lat[~np.isnan(lat)]
+    lon = lon[~np.isnan(lon)]
+
     sin_lat = np.sin(lat)
     cos_lat = np.cos(lat)
     cos_d_lon = np.cos(lon - anchor_lon)
```

The NaN's in attitude are valid and fixed in https://github.com/PX4/flight_review/pull/140.