PX4/flight_review #183: using nginx as proxy server

lijieamd, 2020-01-27:
I already have a web on my server, so i use nginx to proxy http request for flight_review.
everything seems fine but stuck at "Loading Plots..." after ulg file uploaded.
i also tried using flight_review server alone, it works fine.
i don't know much about computer network, hope someone can help.

here is my nginx config and flight_review config:
flight_review startup: 
`./serve.py --address=127.0.0.1 --port=5006 --allow-websocket-origin=*`
nginx config:
```
upstream px4_flight_review {
        server 127.0.0.1:5006;
        }

server {
        listen  80;
        server_name     px4.mysite.com;
        location / {
                proxy_pass      http://px4_flight_review;

                proxy_http_version 1.1;
                proxy_set_header Upgrade $http_upgrade;
                proxy_set_header Connection "upgrade";
                }
        }
```

bkueng, 2020-01-28:
You need to make sure the websockets get through. You can try this:
`./serve.py --address=127.0.0.1 --port=5006 --allow-websocket-origin='*' --use-xheaders`

lijieamd, 2020-01-30:
thanks for your reply, but use `--use-xheaders` option not work
i believe the problem is bokeh websockets  does not proxy right, i'll try some nginx config in Bokeh user guide

malavika-tii, 2024-09-18:
Hi, what was the solution to get this working?

bkueng, 2024-09-24:
This is the nginx config that should work:
```
    location / {
        proxy_pass http://127.0.0.1:5100;
        proxy_set_header Upgrade $http_upgrade;
        proxy_set_header Connection "upgrade";
        proxy_http_version 1.1;
        proxy_set_header X-Forwarded-Proto $scheme;
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
        proxy_set_header Host $host:$server_port;
        proxy_buffering off;
    }
```

malavika-tii, 2024-09-24:
Hi @bkueng, also regarding the bokeh plots?

bkueng, 2024-09-25:
Did you test it? Does it work without nginx proxy?

malavika-tii, 2024-10-03:
Hi @bkueng , should we start nginx after using this config ?

bkueng, 2024-10-08:
You will need to consult the nginx documentation for these specifics.

malavika-tii, 2024-10-08:
> EXTERNAL EMAIL - This email was sent by a person from outside your organization. Exercise caution when clicking links, opening attachments or taking further action, before validating its authenticity.

Hi Beat,

Thanks for your replies. The issue was with respect to the sql db, which was not installed properly. After installing it properly, the issue was resolved. Thanks for your time.

Best Regards,

[cid:9729bba0-a78a-4c85-be65-de2dfdb70243]

Malavika Balakrishnan

Senior Associate - Software Engineer SSRC

PO Box: 9639
Masdar City, Abu Dhabi, UAE

M: +971 50 183 4760
E: ***@***.***

tii.ae<https://tii.ae/>

bkueng, 2024-10-08:
Ok good to know it works now. So #292 can be closed as well?

closed 2024-10-08 by bkueng