PX4/flight_review #187: dbinfo request time out

gulbulut, 2020-03-23:
Hi,

I want to pull all flight logs using the script download_logs.py but I get the error I detailed below. I'm getting time out in dbinfo. what is the solution to this?

`requests.exceptions.ConnectionError: HTTPSConnectionPool(host='review.px4.io', port=443): Max retries exceeded with url: /dbinfo (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f6c365ae490>: Failed to establish a new connection: [Errno -2] Name or service not known'))`

bkueng, 2020-03-24:
Hi

This is a connection error, and it works for me. Can you access https://logs.px4.io?

Turokhunter, 2022-02-11:
Building on this. I'm getting the same error. Server request failed. I can access https://logs.px4.io/ but accessing https://review.px4.io/dbinfo leads to a  504 Gateway Time-out.

bkueng, 2022-02-14:
It should work now, just the caching that took a bit longer.

Turokhunter, 2022-02-14:
Thanks for your help. I ended up rewriting my own dbinfo grabber by taking advantage of https://review.px4.io/browse_data_retrieval. But, I'm sure this will be helpful to others in the future.

timothyZimmermann, 2022-03-02:
Hi,
I am facing a similar issue to Turokhunter. I can access https://logs.px4.io/ but accessing the dbinfo results in 502 Bad Gateway.

bkueng, 2024-12-16:
Closing, let me know if this is still an issue

closed 2024-12-16 by bkueng