PX4/flight_review #189: Airspeed plot disappeared for PX4 v1.10 (and earlier)

ThomasRigi, 2020-04-14:
Recently the plot for airspeed / groundspeed has disappeared on flightlogs using PX4 v1.10.2 and older. Here's a sample SITL log (flown with Gazebo's standard VTOL on v1.10.2) : 
https://logs.px4.io/plot_app?log=7ea35ca1-e508-40c1-85be-5497b472b978

I believe it is this commit that introduced changes on which airspeed topic is plotted: https://github.com/PX4/flight_review/commit/bda9fc9e174fbf17bc656c48b20f4b2bda2c486c
But I don't understand why it doesn't work. L343-346 should keep the compatibility I'd thought.

@jinchengde @bkueng can you look into this please? Thanks :)

jinchengde, 2020-04-14:
It looks like there is no "airspeed validated" topic in old version of log and crashed

ThomasRigi, 2020-04-14:
To my knowledge the "airspeed validated" topic only got introduced after v1.10.

When there is no "airspeed validated" topic, shouldn't the if clause at https://github.com/PX4/flight_review/blob/eaad23cb7f6e51cde73914a2347c8c84d8060460/plot_app/configured_plots.py#L334 trigger the else clause at https://github.com/PX4/flight_review/blob/eaad23cb7f6e51cde73914a2347c8c84d8060460/plot_app/configured_plots.py#L343-L346 ? The "airspeed" topic is present in the old version. Still, there is no plot shown.

jinchengde, 2020-04-15:
@ThomasRigi thanks to figure out this issue

George-avy, 2020-07-13:
Hi! For some reason this fix doesn't work for me.
Even though my log contains the new ``airspeed_validated`` topic, ``indicated_airspeed`` is plotted, instead of ``equivalent_airspeed``.
Weird...

Log: https://logs.px4.io/plot_app?log=4576ddfd-4cd4-4c63-a6b4-77df451bcc54

bkueng, 2020-07-15:
Yeah the check was simply incorrect. Fixed in https://github.com/PX4/flight_review/commit/8112a2285f35af4b710e60be7bf2237b4a2ce5e9.

closed 2020-07-15 by bkueng

George-avy, 2020-07-15:
Ah, nice. Thanks!