PX4/flight_review #200: Add warning GPS Jamming

MartinStokelj, 2020-09-02:
According to [link](https://docs.px4.io/v1.9.0/en/log/flight_review.html)

GPS Noise & Jamming
The GPS Noise & Jamming plot is useful to check for GPS signal interferences and jamming. 
The jamming indicator should be around or below 40. Values around 80 or higher are too high and the setup must be inspected.

It would be nice that graph shows some kind of warning that values are to hight.

dagar, 2020-10-27:
Can you reference the documentation for the relevant receiver?

MartinStokelj, 2020-10-27:
We are using uBlox NEO-M8N modul core of GNSS receiver made by our company.
On some drones we see this bellow 40 on some we have 250. For this with 250 we made shielding on cables and now we are around 150. This is high when we connect IP camera. Without ethernet connection cables the noise is low.

dagar, 2020-10-27:
> We are using uBlox NEO-M8N modul core of GNSS receiver made by our company.
> On some drones we see this bellow 40 on some we have 250. For this with 250 we made shielding on cables and now we are around 150. This is high when we connect IP camera. Without ethernet connection cables the noise is low.

My interpretation was you can only interpret it as a relative metric. From the ublox 8 protocol specification.

[image]

MartinStokelj, 2020-10-28:
thanks for this info.
I think it would be a good feature if graphs shows some kind of warning if values are to high, for example: 
- to high GPS noise
- to high vibrations