PX4/flight_review #206: Automate image build for docker-compose

bbworld1, 2021-02-15:
The px4flightreview image does not exist until you build it manually, meaning the docker-compose config will not launch until you build the Dockerfile. Combined with a relative lack of documentation, this leads to user confusion.

It would be best to add a `build:` section in the docker-compose configuration, which will automatically build and tag the image when the configuration is run.

Side note: have we considered pushing an actual official px4flightreview image to Docker Hub? It may be useful to have.

bkueng, 2021-02-16:
Hi
Feel free to suggest any changes you'd like to see via PR. Personally I don't use docker for this.

bbworld1, 2021-02-16:
Thanks, I've added it at #207 here.

Docker Hub pushes are trivial to automate now that docker-compose handles building (simply `docker-compose build`, then `docker-compose push`), if that's an option you want to explore.

bkueng, 2024-12-16:
You're welcome to add more automation, I'm currently not making use of it though.

closed 2024-12-16 by bkueng