PX4/flight_review #208: Browse tap and satistic app got error

luvwinnie, 2021-03-11:
I have upload an image then i try to tap the browse tap it show nothing, then i try to tap the satistic tap it shows the below errors.

```
srv_1  | ERROR:bokeh.application.application:Error running application handler <bokeh.application.handlers.directory.DirectoryHandler object at 0x7f793c05f320>: list index out of range
srv_1  | File "statistics_plots.py", line 514, in _plot_public_data_statistics:
srv_1  | prev_version = versions[0] Traceback (most recent call last):
srv_1  |   File "/usr/local/lib/python3.6/dist-packages/bokeh/application/handlers/code_runner.py", line 197, in run
srv_1  |     exec(self._code, module.__dict__)
srv_1  |   File "/opt/service/plot_app/main.py", line 65, in <module>
srv_1  |     p = statistics.plot_public_airframe_statistics()
srv_1  |   File "/opt/service/plot_app/statistics_plots.py", line 420, in plot_public_airframe_statistics
srv_1  |     self._all_airframes, 'airframes', 'Airframe', label_callback)
srv_1  |   File "/opt/service/plot_app/statistics_plots.py", line 514, in _plot_public_data_statistics
srv_1  |     prev_version = versions[0]
srv_1  | IndexError: list index out of range
srv_1  |
```

bkueng, 2021-03-11:
Looks like it expects at least some logs. Can you fix it?

luvwinnie, 2021-03-11:
@bkueng Thank you for reply, I'm trying to fix this, It seems like the Bokeh Application(handler) load the plot_app first time it load version nothing, I'm not very familiar with bokeh and tornado, does the Bokeh reload the app?

luvwinnie, 2021-03-11:
it seems like the config of the sqlite data path not correct when using docker, this repository need to be more organize for using docker, I would like to make a PR when I'm the problem .

luvwinnie, 2021-03-11:
I have open a PR #209 , It is the problem when there is no public logs.

luvwinnie, 2021-03-26:
closed issue which fix by PR #209

closed 2021-03-26 by luvwinnie