PX4/flight_review #212: Plots hard to analyze for colorblind people

marcojob, 2021-03-28:
For colorblind people it's almost impossible to distinguish red and green plots in certain cases; would be nice if there was a way to change all plots color to a different set. Example where it's hard to see the difference between lines:
[image]

bkueng, 2021-03-29:
Can you make a specific suggestion?

marcojob, 2021-03-29:
Sure, there is plenty of colorblind friendly palettes, i.e:
[image]
I think it could be a nice addition for the enhancement backlog (if this exists) since around 8% of men and 0.5% of women are affected by some degree of color blindness.

bkueng, 2021-03-30:
Sure, I have no problem changing colors, it's just that everything needs to be checked to still work and harmonize.
I can do that when I find the time.

KarthiAru, 2024-07-26:
@bkueng Any update on this? I'm colorblind too and find it hard to distinguish red and green lines. 
Where are the config for the color codes specified? In the meantime, I can make the changes in my local server.

KarthiAru, 2024-07-26:
I updated the colors in config.py in my local based on the palette shared by @marcojob. The combination of colors also matters. I had to use a combination of a darker & a lighter color for the two primary colors to distinguish better.

[image]

```
# Vermilion: #d55e00
# Yellow: #f0e442
# Blue: #0072b2
# Blue Green: #009e73
# Orange: #e69f00
# Sky Blue: #56b4e9
# Pale Violet: #cc79a7
# Black: #000000

plot_color_blue = '#56b4e9' # original: #2877a2, #3539e0
plot_color_red = '#d55e00' # original: #e0212d

color_gray = '#464646'
colors8 = ['#d55e00','#f0e442','#0072b2','#009e73','#f0e442','#56b4e9','#cc79a7','#000000']
colors3 = colors8[0:3]
colors2 = colors8[0:2] # for data to express: 'what it is' and 'what it should be'
```

closed 2024-08-14 by bkueng