PX4/flight_review #218: Unable to upload log file *.ulg - HTTP Error 400: Invalid File

MuhammadBilal1, 2021-06-05:
I am trying to upload the flightlog files to https://logs.px4.io/ and getting an error message all the times.

Tried multiple files but the error is same.

File size: 1.18MB
FileName: log_43_2021-6-5-16-13-40.ulg

Is there any other way to open this log file.

[image]

bkueng, 2021-06-08:
You can test if it works locally:
```
pip install pyulog
ulog_info log_43_2021-6-5-16-13-40.ulg
```

MuhammadBilal1, 2021-06-08:
@bkueng Thanks for your response. I didn't tried your solution yet.
But I copied the files directly from the memory card of PixHawk and then uploaded them, it worked.