PX4/flight_review #219: 500 Internal Server Error when uploading .ULG

FlyingFish136, 2021-06-15:
I am totally new to GitHub and have absolutely zero coding experience... However, for my Masters thesis project we have used a PX4 inside a light aircraft to record the GPS and telemetry data.

When I go to upload a .ulg file I get sent to this web address and given this server error - any ideas?

https://logs.px4.io/upload

[image]

bkueng, 2021-06-15:
Hi. Can you provide the log file? You can also PM me if you prefer.

FlyingFish136, 2021-06-15:
I can't seem to upload the file for some reason. I even sent it to a compressed ZIP, is this a GitHub thing I'm unaware of? I made a copy of the file and opened it in Notepad++ where as far as I can tell it has recorded data, whether or not it's all garbage I of course don't know. Shall I email you the file @bkueng? :)

bkueng, 2021-06-16:
Yes. But first try again to upload again, there was actually an issue on the server.

closed 2023-06-19 by bkueng