PX4/flight_review #222: Crazy large logging duration and flight time values affecting graph axes limits

czasz, 2021-07-28:
I am trying to analyze my flight data but the flight review tool shows astronomically large values for both the logging duration and flight time which in turn affects the graphs and makes it very difficult to analyze the data.

This is the first issue I see (the actual flight time is about 5 minutes):
[image]

This causes the graphs to look like this, which is very clearly driven by the incorrect log duration values:
[image]

However when I zoom in a crazy amount, everything looks fine:
[image]

Does anyone know the cause of this issue/bug? It looks like it's a timestamp issue on PX4's side but from what I can see there is an extremely small number of parameters that affects the logging system.

bkueng, 2021-07-29:
This generally happens when a timestamp is invalid. So on the PX4 side. You can look at the raw data to find the topic (convert to csv or instrument pyulog).

czasz, 2021-07-31:
Thanks @bkueng that was indeed the issue, I dug into the firmware source code and was able to find the problem.

closed 2021-07-31 by czasz

bkueng, 2021-08-02:
I take it that was something in your code, and not upstream?

czasz, 2021-08-03:
Yes that's correct