PX4/flight_review #223: Linking time axes between plots

uavfactory, 2021-10-18:
Hello all,

Sorry, this is actually a feature request. I think it would be quite handy if Flight Rewiev had a possibility to link the time axes of the all or selected plots with a check box. I always use this feature on Plot Juggler.

Best,
Ozan

junwoo091400, 2022-12-23:
Seems like this is probably doable with: https://docs.bokeh.org/en/latest/docs/user_guide/interaction/linking.html

junwoo091400, 2022-12-23:
Seems like this was actually removed intentionally from this commit: https://github.com/PX4/flight_review/commit/6851ec230ec9dec381e9ad49e19ffc2af2c8ae6a#diff-b21808fb881ff9377c69f652e4f30d0dd75598f6bf55f1f9469b81d35b662c83

" plots: make sure the initial x-axis range is equal for all plots

otherwise it can be confusing if some message stops being logged/published
at some point."

junwoo091400, 2022-12-23:
@uavfactory I was able to implement this quickly using this commit: https://github.com/junwoo091400/flight_review/commit/3825ccca59cea2d1323daec194171f5feb811291

But I noticed a significant decrease in responsiveness of the graph when I am pan/zooming them.

I am wondering what would be the best way to reduce lag, but also provide flexibility in providing synchronized plot (as I think this is really useful, as used also in the PlotJuggler, as you mentioned)

[image]
[image]