PX4/flight_review #226: 3D View Altitude Source

uavfactory, 2022-01-05:
Dear all,

It seems that 3D View replaying takes the GPS altitude to plot the height. However, as GPS data is less frequently updated, the rflight looks climbing stairs. 

My solution offer is that, it should be plotted with "Fused Altitude Estimation"

[image]

Best,
Ozan

bkueng, 2022-01-10:
It's deliberatly using the GPS as it's more reliable on a crash.
Maybe an option to select could make sense.

uavfactory, 2022-02-01:
> It's deliberatly using the GPS as it's more reliable on a crash. Maybe an option to select could make sense.

Dear @bkueng,

Thanks for your reply! Could you add this option in the upcoming release?

bkueng, 2022-02-02:
I might, but you're certainly welcome to contribute that.

uavfactory, 2022-02-03:
@bkueng 

I think ideal plot should be as below. By this way, there won't be a need for additional button to activate either of the modes.

- Position estimate: Smooth continuous yellow line
- Raw GPS position: Discrete dots