PX4/flight_review #230: Public log file download issue

abhishekgs1, 2022-03-12:
Hi 
I am unable to download the public logfile from the code provided 
It seems that DB API is not responding
Can you please check and update if I can download the files from the script

Please find the error attached below
[image]

Regards
Abhishek

bkueng, 2022-03-14:
Can you try again? The cache update takes a bit longer sometimes.

abhishekgs1, 2022-03-16:
Hi I tried but still the error exists, also I tried to ping https://review.px4.io/dbinfo which is part of the code still I got the bad gateway error

[image]
[image]

abhishekgs1, 2022-03-16:
Do you have any suggestions or other code to get this data?
if yes please share

abhishekgs1, 2022-03-16:
> Can you try again? The cache update takes a bit longer sometimes.

bkueng, 2022-03-18:
I cleaned up the db a bit, it's working for me again.

bkueng, 2023-06-21:
Fixed in https://github.com/PX4/flight_review/commit/62b75900b78106b13e6f04faf7e1b5d37e117570

closed 2023-06-21 by bkueng