PX4/flight_review #233: Scipy install is broken

eduardkieser, 2022-04-05:
Running `pip install -r requirements.txt` in a new venv (tested on python 3.10.0 and 3.9.5) causes some gcc compiler error, looks like it's compiling numpy for scipy 1.3.3. It spams a few 1000 lines from gcc and ends with:
```
error: legacy-install-failure
      
      × Encountered error while trying to install package.
      ╰─> numpy
      
      note: This is an issue with the package mentioned above, not pip.
      hint: See above for output from the failure.
      [end of output]
```
Removing the version pin for scipy seems to work (but obveously might cause other problems).

bkueng, 2022-04-06:
We can remove the version, that should be fine, I've seen such problems as well. scipy is mainly used for the spectrogram plots.

mengchaoheng, 2022-04-20:
sudo apt install python3-scipy

bkueng, 2023-06-19:
This was changed in https://github.com/PX4/flight_review/pull/249/files

closed 2023-06-19 by bkueng